tomek-szczesny/OpenUPS #3: Battery - UPS interconnection

tomek-szczesny, 2023-04-23:
For now I've added XT60PW-M as a schematic placeholder until we settle on the target PCB interconnection.

tomek-szczesny, 2023-06-06:
Can't believe this is still an issue. ;)

I just added a screenshot of the current state of PCB design on Odroid forum.
I'm not diving into a PCB design for another few days so I'll take this opportunity to discuss the possible improvement in terms of the PCB layout.

Let's assume the upper edge on this drawing is the front side of UPS in mini case. Where USB-C and LEDs are.

[image]

Long story short, I'd like to discuss the possibility of placing all three battery connectors on the front: Heavy lifting XT30, two-pin JST for pack sense and Qwiic for temp sensor I2C.

[image]

The obvious drawback is having all these connectors on the front of the device. I can't tell how much worse this is than having them on the side, the question is open to the broader public of potential users.
And this solution assumes that some connections between both PCBs of a drivebay build will be discrete, using separate wires. Personally I don't see it as a problem, given how much time we've already invested seeking for a suitable board to board connection.

There are a few reasons for this layout:
- All these connectors and BMS work with a "Different GND", or BATT-. With BMS fitted somewhere in that corner, it will minimize the area of foreign voltage domain. Note that GND is also our primary heatsink, so we want to maximize it.
- Less footprints, more space. With one XT30 less we can actually place Qwiic on top of it.
- Bottom side XT30 and JST pin holes will be extended with wire pads on the bottom side. Same for BAT PCB that may have the same pattern for the same purpose. That is, if we are willing to accept soldering four wires in drivebay build. Heck, we can as well go all in and do all the 7 connections as discrete wires while we're at it.
- I2C is closer to AVR, and I2C isolator may be placed away from the BMS (next to LEDs)
- More space for charger heatsinking pours.

hominoids, 2023-06-07:
Replacing the inward facing XT30 for soldered wires is not a big deal but moving the other XT30 to the front really undermines it's purpose.  If it has to be done I understand but I would then advocate to remove it entirely because there would no use case for it in the front and using a pigtail feed through the back would be the better and preferred battery connection IMO.

tomek-szczesny, 2023-06-08:
Thanks for your input, that's exactly why I'm asking. Sometimes I know we had decided something but can't remember why.

Here are two new options to consider. Both are almost the same. Single XT30, 2-pin JST and Qwiic are placed on the right hand side. Two bottom side pads, 5x10mm, can accept very thick wires for drivebay build.
The difference is in 6-pin mixed I2C and bat sense connector, dedicated to drivebay build. Either a THT bottom side JST, or top side SMD FPC ribbon with 1mm pitch (TE 84953-6).
Six pin link would be "4S ready".

I think ribbon makes more sense due to its high flexibility, should be easy to hook up between both PCBs even after they are mounted in the drivebay box. 
The downside of this solution I can see for now are XT30 and wire pads overlapping on the bottom side. It's not a big issue considering two overlapping XT30s were a base solution we're trying to improve now.

[image]
[image]

hominoids, 2023-06-08:
I like both  solutions and can live with either one.   IMO the JST connector makes a little more sense then the ribbon cable for a couple reasons.  The first is that they are tougher and can take more abuse and are equally easy to hookup.  When silicon insulated wire is used they may be more flexible then a ribbon cable and can be acquired anywhere and/or easily made to any length needed on the fly.  The JST could also be used to extend the standalone case to support 4S, or use the combined connector if preferred, by soldering wires to the pads and running a pigtail through a side or rear case wire slot.  I cannot tell the size of JST your proposing but a JST-PH(2mm) would be my choice. FYI, I have a 100' roll of high strand silicon insulated 24ga wire that was incorrectly sent to me that could be used to make JST cables once  I order up a larger JST connector kit(5-8pin).

tomek-szczesny, 2023-06-08:
OK so how about this. A unified 6-pin connector for standalone and drivebay battery packs in JST-PH format. Two footprints, horizontal on the bottom side (purple) and vertical on the top side (teal) share the same holes. Above I've drawn an example PCB layout on BAT PCB, with vertical and horizontal SMD JST footprints (horizontal being closer to the edge). As far as I can remember we gave up XT30 on BAT PCB at some point and decided to have it pigtailed.

That is also the most space efficient solution to date. The question is, can we afford the top side vertical JST connector on a UPS PCB?

I would appreciate if you took over the JST cable making duty. :)

[image]

hominoids, 2023-06-08:
That sounds good, I like the solution.  FYI, the [last discussion](https://forum.odroid.com/viewtopic.php?f=55&t=46481&start=151) about the battery connections had a rear facing and vertical XT30 with an additional solder pad set.  The rear facing was for 3S2P drive bay case mounting, the vertical was for standalone and drive bay case use, as were the solder pads.  Now, that all got set aside IMO when we changed and moved the BMS to the UPS PCB.  Since there is plenty of room the question now is what to include on the battery PCB and where.  I don't see any reason we cannot keep the same configuration but I'm open to other suggestions.  You can also view 'all connectors' in the OpenScad model to visualize the current setup.

tomek-szczesny, 2023-06-15:
As far as I can remember, in every drivebay configuration our top concern was height of a battery pack. We slimmed down PCB to 1mm and even the case floor where PCB is supposed to be, just to barely fit into the drivebay format height.
There are no SMD XT30 connectors on the market, I believe, and we can't use THTs in a drivebay battery pack, because it has to lie flat on the case bottom.

Digesting all these facts, in my mind, the conclusion is this: We can add XT30 connectors to battery PCBs, but only if the cell size will be limited to 18650, or battery case size enlarged over drivebay specs. 

XT30 is not needed for integrated drivebay OpenUPS build because solder pads make more sense in this build.

3S2P battery pack, if we think about it, doesn't need to fit into a drive bay, because the standalone OpenUPS in mini case won't fit the drive bay either. It will be a standalone modular device, without any claims of fitting anywhere in particular. However it would be nice if they were stackable in an elegant fashion.

So here's what I propose:
- OpenUPS in drivebay case as planned before, interconnected with short 6-pin JST cable and two soldered wires
- OpenUPS mini, with XT30 input and 6-pin JST connector on its side,
- Standalone battery packs: 3S1P and 3S2P, the latter having two PCBs inside. Both should have XT30 and JST exposed... perhaps on their side?

I forgot how much fun sketching is. This is my concept art of standalone OpenUPS. :)
There is nothing against 3S4P at this point, if someone really likes the idea. The only limit are 8 thermal sensors, and I foresee one for each 3S1P PCB.

[image]

hominoids, 2023-06-15:
That sounds good.  I have not started work on the dual PCB 3S2P case because I'm waiting for the layout to settle more.  I'm not worried about it because the current configuration and sense line setup allows for any number of different battery pack configurations.  Once we have a finalized layout I'll put the time into figuring out other case possibilities.

tomek-szczesny, 2023-06-18:
Could you help me determine the "correct" XT30 pinout? The footprint has plus and minus marks on a silkscreen layer but I'm not convinced they are correct. This is a male XT30 on a bottom side of a PCB (as seen from top). I heard the actual connectors have polarity markings on them, which would be nice to follow.

hominoids, 2023-06-18:
That appears to be correct.  The + side is on the flat side of the connector and the - is on the keyed side as shown in the photo of XT60's which have the same orientation.  Ignore the wire position in the photo, it's wrong.
[image]

hominoids, 2023-06-18:
I just noticed there are two plus symbols so let me clarify, I believe the plus side should be at the square pad position.

tomek-szczesny, 2023-06-20:
Hmm my previous screenshot must have been confusing. It was showing the bottom side silkscreen, but seen from the top side. 
Here are two renders of the same spot with no changes. As long as your 3D model is correct, and indeed flat side is a plus, then I think the square pad should be a minus. 
Which is also in line with a Kicad built-in XT30 footprint.

[image]
[image]

hominoids, 2023-06-20:
That is correct as indicted in the picture below.  The XT30 minus mark can be seen on the keyed side attached to the black wire and the red wire is attached to the flat side.  I think your good.
[image]