tomek-szczesny/OpenUPS #11: Use THT electrolytic caps for easier repairs

tomek-szczesny, 2023-05-03:
As stated in the title, I know exactly how hard it is to remove those SMD cans, and solder in new ones on a crowded PCB.

@hominoids, what is a safe estimate of a UPS PCB top side clearance? Can we use 11mm tall capacitors?

hominoids, 2023-05-03:
The red rod is 11mm tall so an 11mm tall capacitors will fit without an issue.  
[image]

tomek-szczesny, 2023-05-07:
This has been decided that we'll use THT electrolytic caps up to 7mm high, on a bottom side of UPS PCB, where applicable.

tomek-szczesny, 2023-05-10:
Seems like there is an awful lot of caps fitting into 6.3mm diameter, 7mm height, 2.5mm pin spacing. Here is one I have selected for Vbus and 12V:
https://www.mouser.pl/ProductDetail/Rubycon/25ZLG47MEFC6.3X7?qs=T3oQrply3y99xHUW5IeoPQ%3D%3D

The 1000h life expectancy is specified for 105C ambient temperature, maximum voltage and constant current cycling. After that time the capacitor is not dead, just degraded. At 45C ambient temperature alone, we should be expecting 2^6 times longer life, which is about 7 years. In practice, longer than that due to gentler treatment they will be experiencing.
Source: https://www.xppower.com/resources/blog/electrolytic-capacitor-lifetime-in-power-supplies

closed 2023-05-17 by tomek-szczesny