tomek-szczesny/OpenUPS #12: Polyfuses for each output

tomek-szczesny, 2023-05-03:
I'm thinking about:
- 5V: 4A
- 12V: 5A
- Vadj: 4A
- USB 5V: 4A
- SATA: No fuses. After all, motherboards don't have special fuses for SATA drives.

tomek-szczesny, 2023-05-05:
So far I've implemented:
- PTS18126V260 for 5V output (screw terminal). Passes 5A min, trips at 8A in 2.5s.
- 0ZCF0200AF2C for 12V and Adj outputs. Passes 4A, trips at 8A in 5s.

I don't like the latter, it's huge.

USB will be protected by current limiting switch.

tomek-szczesny, 2023-05-16:
Studying PTH fuses in greater detail, they do not respond fast enough, and their current rating is not just a single number that says it all. 5A fuse is guaranteed to pass 5A, but will act on 25A after 2s, for example.

After some more thought, I'm thinking all power converters shall assure sane current output limits, each according to its capabilities. Fuses are supposed to protect outputs against converter fault, mostly. Thus, I think blow-fuses are justified here. 

https://www.tme.eu/en/katalog/smd-fuses_100131/?params=2%3A292%3B617%3A161c40%3B10%3A15ec4e%3B35%3A15ec66%3B247%3A15fa8d%3B2629%3A161c45&onlyInStock=1

At the same time, we don't want to pop the fuses during "nominal working conditions", which also includes end user doing short circuit. So it may make sense to use time-lag fuses:
https://www.tme.eu/en/katalog/smd-fuses_100131/?params=2%3A292%3B617%3A161c40%3B10%3A15ec4e%3B35%3A15ec66%3B247%3A15fa8d%3B2629%3A161c44&onlyInStock=1

1206 pads will let us pick any fuse we like for each output.

tomek-szczesny, 2023-05-17:
The solution that I just adapted will surely blow fuses due to output capacitor discharge on short circuit.
So it makes sense to move fuses between converter output and capacitor bank.
All converters support controlled ramp-up so fuses shouldn't blow on start-up. 
Probably a good idea then. There's no reason not to use fast acting fuses now.

tomek-szczesny, 2023-05-18:
It gets complicated because we have multiple outputs. Let's consider this scenario:
```
Converter --> fuse --> caps +--> 5V screw terminal output
                            +--> 5V SATA output
                            +--> USB switch with current limit --> 5V USB output
```

Screw terminal output and SATA output are not protected against shorts, shorting one will affect all 5V outputs. This is also true for 12V output. On the other hand, that is to be expected from a multi-voltage computer PSU.

I think this is acceptable, as only USB is considered a "user accessible output" for daily use. All the remaining ports should be treated as semi-permanent. Just like typical computer architectures do.

tomek-szczesny, 2023-06-29:
Looks like we can afford eFuses after all:
https://www.mouser.pl/c/?q=TPS25947
Some $1.5 for an adjustable FET-based fuse, with digital signaling and so on. 

Those could be used directly on screw terminal outputs, regardless of capacitor banks and so on.

USB got its own protection, and other misc outputs (fans, SATA) can get blow fuses or PTCs, or no protection at all. Taking Odroid H3 schematic as a reference, they have no overcurrent protection on fan or SATA outputs. So probably SMD blow fuses will be more than adequate.

f1vefour, 2023-07-01:
What is the function of Vadj?

tomek-szczesny, 2023-07-02:
> What is the function of Vadj?

Adjustable (programmable) output.