tomek-szczesny/OpenUPS #13: USB PD output

tomek-szczesny, 2023-05-03:
Since we've committed to USB-PD output, it'd be nice to actually implement it at some point.

The problem is, each user may have different expectations for this port. Someone may mistake it for a power input. Others may wish to charge their laptops off it. All compliant devices require special signalling on CC pins to source current from USB-C, and conversely, should offer a handshake, otherwise we may run into situation of two chargers playing tug of war. 
There must be some chip or another to advertise supported modes of operation and avoid damage.

Ideally we want a chip that does all that: Advertise voltage and power options, limited to some sane value like 30W, and drives a switch or a built-in power converter.

One solid candidate is [TPS25740](https://www.ti.com/lit/ds/symlink/tps25740.pdf), although it seems that it must be accompanied by a separate voltage converter it could control directly. This may be arranged, but again, cost, space, all that stuff.

We could limit the scope of USB PD to 5V output only (that was the original goal), but I can't find chips with such limited scope.
The only suitable chip for this task we've identified so far is [IP6518](http://www.injoinic.com/wwwroot/uploads/files/20200220/c55f78d32299df716a7c737ce28c2af2.pdf) that is hard to source outside Sinoverse. A Western equivalent would be literally perfect.

To wrap it up:
- We need a chip that can do the talking via DP, DM, CCs
- A solution for generating required output voltages (optionally, we've got 5V source already and 5V-only mode is also fine)
- We've got 5V and 12V high-current sources at our disposal, and 9-24V unregulated rail

hominoids, 2023-05-03:
Something similar to the Ti chip is the [onsemi FUSB15101MNTWG](https://www.digikey.com/en/products/detail/onsemi/FUSB15101MNTWG/17053360).

This might be the perfect solution though since it appears to be fully self contained, just add power, [Infineon CYPD3177-24LQXQT](https://www.digikey.com/en/products/detail/infineon-technologies/CYPD3177-24LQXQT/10238323)

tomek-szczesny, 2023-05-04:
Good grief, my head will explode from all this new information.
[Here](http://ww1.microchip.com/downloads/en/appnotes/00001953a.pdf) is a a very nice Microchip paper on USB-C, which goes into the details of how exactly CC pins work.

Long story short, PD is not mandatory and not needed for what we're trying to achieve.

So this is what happens:
- DFPs (Downstream Facing Ports) advertise their current capability by pulling up both CC pins. For 5V 3A output, that's 10k pull-up to 5V (Table 6).
- 3A is the maximum VBUS current without DP
- UFPs implement pull-down resistors (or other means) in order to complete a voltage divider network
- Cable Attach Detection is a mandatory feature that works by sensing CC pin voltage
- If 3A is advertised, CC voltage under 2.75V means the connection is established and valid.
- Both CC pins must be monitored, and the results OR'ed (voltage drop on one pin is enough)
- UFP (The device) is responsible for limiting its current draw, not the source
- That means we may as well offer up to 5A for non-compliant devices, and still be compliant with USB specs.
- "5V to VBUS may only be applied when a valid cable attachment is detected. This prevents two downstream facing
ports from back-driving current into each other."

So if my understanding is correct, all we need is:
- CC pins pull-ups
- Two comaparators with wired-OR that monitor CC voltages
- a PMOS switch (or dedicated USB load switch).
...and we should be all set.

tomek-szczesny, 2023-05-04:
Ah, here's a good candidate: A dedicated USB switch, 3A (limit above 3.5A), active low enable signal, $.50:
https://www.mouser.pl/ProductDetail/Diodes-Incorporated/AP22814BW5-7?qs=lqAf%2FiVYw9gFAqm%252B3YvdqA%3D%3D

Unless we do want to support those non-standard output currents:
4A: https://www.mouser.pl/ProductDetail/Diodes-Incorporated/AP22800HB-7?qs=sGAEpiMZZMv0DJfhVcWlK1Yz3z1OExkkqgcb3xeQGB45NBTOkdirIQ%3D%3D

closed 2023-05-10 by tomek-szczesny