tomek-szczesny/OpenUPS #14: Fuel Gauge

tomek-szczesny, 2023-05-04:
Fuel Gauge chips are expensive, and the market is dominated by ICs for single cell systems.
This is not a feature that is in demand or necessary, just leaving it here as a possible improvement for the future.

A crude fuel gauge could be implemented on AVR (that already monitors the battery current).
But it should not be attempted at least until the current measurements are proven to be reliable enough.

paulcrawford, 2023-05-04:
My data collector system that runs in C on my N2L does complete management of an LiFePO4 battery (24V 70Ah) including charging and load monitoring both voltage and current that runs a 750W three phase motor via a 240VAC inverter and single phase to three phase VFD. It includes individual cell protection for low voltage and high voltage and temperature. I use 2 industrial grade current shunts to measure battery output and input in addition to charging current from a 240VAC charger specifically for LiFePO4 batteries.

The above is of course a much larger scale of power than OpenUPS but here is an example for 10A:

https://canada.newark.com/canadian-shunt/la-10-100/base-mounted-dc-shunt-100mv-10a/dp/73K4798?pf_custSiteRedirect=true

It is of course very bulky but also very accurate.

The software I have developed does not care about the scale of power being monitored so perhaps could be adapted to the OpenUPS AVR MCU.

tomek-szczesny, 2023-05-04:
Battery current monitoring is already implemented and will be available directly at AVR ADC input.

[image]

Its range includes 5A charging and 15A discharge current. It is deliberately slowed down by low pass filter to give more meaningful results when probed less often. I'll tune this filter's corner frequency below Nyquist frequency later, so we can reliably integrate current in either direction.

If your software can monitor battery charge and return its status in % or Wh, and require no prior configuration of cell capacity, that would be perfect. Manual battery charge configuration is also possible via I2C commands.

All the safety features are already assured by other mechanisms in the system, but thanks for mentioning them. AVR will have no access to individual cells, only the two battery leads.

paulcrawford, 2023-05-04:
When I started developing my system, about 20 years ago, it was based on flooded lead acid batteries. Then I moved on absorbed glass mat (AGM) batteries, which are still essentially lead acid. A lot of effort went into developing capacity tests that could determine the Peukert constant for the batteries. Knowing that constant allows the calculations for per cent capacity remaining. But as the actual capacity keeps changing over time the capacity test needed to be done to determine the new values.

With the advent of Lithium batteries, the Peukert constant became much lower and predictable enough to not require the sophisticated capacity tests any more and if the cell is of good quality the capacity stated on the label is what you get with much less degradation over time as with lead acid. For my needs, house batteries on a boat, the LiFePO4 variety of Lithium is the best from a safety point of view. @Hominoid did mention possibly using LiFePO4 cells. Has a final decision been on that yet? They do have different characteristics than other LiIon cells but I am sure you have considered that already.

The BQ77915 looks interesting but the maximum voltage and current limits, 22V and 4A are too small for me. Perhaps they make a high power version as well. I use latching relays rated at 50A to cutoff load and charging when required using an independent 8 channel analog to digital voltmeter and 4 DS18B20 temperature sensors to monitor 8 cells. Experience by others shows that with proper initial top balancing there is really not much need for continual balancing for the application I have. The OpenUPS is different and you get balancing already with the BQ so that should be fine.

tomek-szczesny, 2023-05-05:
LiFePO cells were discussed and it has been concluded that those would require a separate, new design of the cell carrier board (different protection mechanism). With their poor output current rating, the functionality of OpenUPS will be limited, but still should work nevertheless, for applications up to around 40W.
6-pack LiFePO4 would make more sense, which is also possible at this point. 

What I was hoping for is a coulomb counter algorithm. A piece of software than can track charge flowing in either direction, and thus learn maximum battery charge, as well as predict its state. The "learning" part is important, as cells come in great variety and condition.

We're only after a very rough estimate of a battery charge. This could enable a user to script their way into shutting down a server when estimated battery charge is below 15%, for example. 

The only benefit of using a dedicated chip is that it has a proprietary algorithm inside it.

tomek-szczesny, 2023-05-10:
The coulomb counter has been included in the battery pack (#1).
Charger circuit (#24) has "Learn mode" which will be used for measuring battery capacity.
I think this is good enough for our purposes.