tomek-szczesny/OpenUPS #16: LED indicators

tomek-szczesny, 2023-05-05:
I keep forgetting about them.

I'm thinking about the following LEDs, driven by MCU:
- green Batt LED: Blinking when charging, solid when charged
- red Batt LED: Blinking when discharging, solid on error (no battery, most likely)
- Green LEDs "power good" for each of three voltage converters: 5V, 12V, Adj
(Even though some power converters have Power Good outputs, I think it should be driven by a microcontroller)

We could add a "power on" LED but that feels redundant - at least one Batt LED will be on anyway.

Now when we need extra 5x GPIO, the real question is whether we switch to 48-pin AVR, or add I2C port extender...

hominoids, 2023-05-05:
"Power on" is probably not needed if there are three "power good" LED.  To avoid increasing pin count the "power good" could be AND together using one LED and the "charging" and "discharging" could be combined too. Off when charged, solid when charging and blinking when discharging.

tomek-szczesny, 2023-05-06:
I made a little search and concluded it's far cheaper to use a bigger AVR than adding a dedicated external LED driver.
By "Bigger AVR" I mean a 48-pin package, which has more pins, but is physically smaller than 28-pin variants.
We already have too few pins at this point, so we may as well commit to this change already.

There is also a way of connecting 6 LEDs to 3 GPIO, which would require time multiplexing to control them, but let's not get THAT cheap.

I suppose we can allow ourselves to provide as many LEDs as we deem necessary and practical.  We don't want to remove any useful indicator, but at the same time we don't want to create an interface that is too cluttered and hard to read.

"Power good" LEDs are meant to show not only that the output voltage is within limits, but also indicate which power outputs are on. I think I'll let the user switch power outputs on and off via I2C commands. On a second thought, there is no good reason to insist on driving these LEDs by MCU, so 5V power good will be driven by 5V Buck directly.

Anything else we wish to notify a user about?

tomek-szczesny, 2023-05-06:
I added 5 LEDs for now, as per the original description.
We still got pins left after I did #20 :)
[image]