tomek-szczesny/OpenUPS #33: Additional mount hole

tomek-szczesny, 2023-05-24:
A fifth hole is to be added somewhere near the connector-filled edge, around the center. 
At the same time, some of the existing holes may be moved for better THT cap placement.

tomek-szczesny, 2023-05-24:
@hominoids, I moved one hole 2mm "up", in order to make room for bottom side capacitors. 
No need to update models on your side until I add the fifth hole (not today).

[image]

hominoids, 2023-05-25:
My plan right now is to wait until your done with the layout before updating the model and drawings.  Make whatever changes you need.

tomek-szczesny, 2023-05-25:
Exactly what I had in mind, I'll carry on accumulating my changes here then.

tomek-szczesny, 2023-05-25:
South east hole moved 3mm down.
[image]

tomek-szczesny, 2023-05-28:
[image]
@hominoids , the fifth hole has landed. Please update your mechanical design. :)

hominoids, 2023-05-29:
I made the changes to date and added an option in the GUI to show the KiCAD Model for alignment verification.  For now it's kept in ./case/lib and can be updated by replacement.

[image]

tomek-szczesny, 2023-05-29:
Looks like spray painted. ;) 
I'm glad the alignment isn't broken yet. Quality documentation paid off.

Let's keep this issue open for future mechanical changes, if any.

hominoids, 2023-06-04:
I was able to increase the UPS PCB from 90x62mm to 96x64mm.  Take notice of the area directly over the drive bay nut holder that has been trimmed away for PCB clearance, no components there.  I also superimposed the old KiCad model to show how the holes have been moved. 

[image]

The forward UPS mount holes cannot be moved any further outward because of the narrower battery PCB.  And, the rear battery PCB holes/standoffs have been moved forward 2mm more to accommodate the extra depth of the UPS PCB.  I'm still working to see if a dual 3S1P case is possible.

DXF Drawing of the new enlarged PCB and holes.

[OpenUPS_PCB_Changes.dxf.zip](https://github.com/tomek-szczesny/OpenUPS/files/11644619/OpenUPS_PCB_Changes.dxf.zip)

hominoids, 2023-06-04:
I added the keep out areas and also made other adjustments so don't use the previous version for anything.  The keep out area for the bottom inward facing XT30 is not in the correct location and needs to be repositioned since it has moved. There is extra room around the rear I2C that could be used for another connector if needed.  It could face inward and serve the battery PCB on another bus or the area could be used to replace one of the JST-SH(1mm) with a JST-PH(2mm) for a cabling choice.

[OpenUPS_PCB_Changes_Updated.dxf.zip](https://github.com/tomek-szczesny/OpenUPS/files/11646375/OpenUPS_PCB_Changes_Updated.dxf.zip)

tomek-szczesny, 2023-06-06:
@hominoids we could use the extra space on a back facing edge to move fan ports apart, so both could support 4-pin connectors. Could you estimate how far away from each other they should be? I assume it's about 10mm and some.

tomek-szczesny, 2023-06-06:
How wide is the inter-PCB overlap? I'll add a keep out zone there as well, as it's been tempting to place capacitors in there.

hominoids, 2023-06-06:
The 4-pin fan connector is 10.5mm wide and the 3-pin measures 8.17mm.  Here is a photo of both so the positioning relative to the friction clip and wiring is clear.

[image]

There is a 10mm overlap between the PCB. Note, currently only the east side has components on the battery PCB so I've added a 10mm x 33mm keep out zone only in that area.

[OpenUPS_PCB_Overlap.dxf.zip](https://github.com/tomek-szczesny/OpenUPS/files/11668029/OpenUPS_PCB_Overlap.dxf.zip)

tomek-szczesny, 2023-06-14:
I've done some sokoban exercises, spreaded out FAN and Qwiic connectors. Those changes will be available in the following commit. Are you able to extract component coordinates on your own?

FAN connectors are 11mm apart (measured between their centers), so two 4-pin fans should be supported now. 

Also by squishing the components in there I made more room for the charger.

[image]

The same commit will include this preliminary placement of 6-pin JSTs, sharing common pin holes, and XT30 which more or less remained in the same spot.

[image]

I'm sorry for the long delay, I'm still recovering from an unpleasant illness.

hominoids, 2023-06-15:
Sounds good, I hope you feel better soon.