tomek-szczesny/OpenUPS #34: New approach to battery supervision

tomek-szczesny, 2023-05-29:
Investigating a new idea of removing (or significantly reducing) battery supervision microcontroller.

The foreseen benefits include single, centralized firmware on UPS AVR, and removing need for support of a middleman in communication. Also, users will not have to push the button on the battery pack PCB to boot it after assembly. And finally, disconnected battery will actually enter a long term storage mode, and will only work with data link connected. 

This is how I see it as of today:
- Internal I2C bus present on UPS PCB gets extended to 2-pin data connector of the battery,
- UPS PCB's AVR can talk to BQ769x0 directly via I2C, and only I2C
- BAT PCB has a mechanism that puts BQ chip into SHIP mode if no I2C activity is detected.

To me it sounds foolproof. It also serves as a protection against AVR failure, a measure unforeseen until now.

SHIP mode may be activated by either sending a special I2C sequence, or forcing the BQ chip to go through POR (Power On Reset). The latter is discouraged by the datasheet.
At the same time, a mechanism that transitions BQ chip from SHIP to NORMAL mode is required. This is achieved by pulling a temperature sensor pin TS1 above 1V threshold for at least 2ms.

The BAT PCB implementation could be a minimal AVR with I2C port and one more GPIO. Attiny 212/412 fits the bill.
During normal operation, it would passively watch SCK and SCL, possibly with its internal TWI peripheral disabled. If it detects no activity (SCK idle for > 1000ms), it will turn on its I2C Master interface and order BQ to go into SHIP mode. AVR will then lose its power source, as I assume BQ's external LDO will be shut down as well (I can't find confirmation for this). If not, AVR will implement sleep mode.

When BAT PCB gets connected to UPS PCB, either AVR will get temporarily powered from the I2C lines, or it will wake up triggered by SCK pulled high. Both of these may be implemented concurrently. It will then wake up BQ chip by pulling its TS1 pin high. From this point AVR will be powered entirely from BQ's LDO.

Alternatively, BQ wake up may be implemented as a I2C command that battery side AVR will detect and execute on demand. 

Finally, the battery side AVR may support up to 3 configuration jumpers for user configuration of battery chemistry type. Those may be readable via I2C. One of these pins may be used for programming said AVR via UPDI. 

Am I missing anything?

hominoids, 2023-05-29:
A diagram... :)
I made some system diagrams to document and better understand the implementation.  The first is the original approach and the second is the new proposed approach.  Where should we permanently store this information?  What needs to be done to add it to the wiki?   I used drawio available [here](https://github.com/jgraph/drawio-desktop/releases/tag/v21.2.8) if you want to directly change either.
[image]
Source file: [OpenUPS_Diagram-v0.90.drawio.zip](https://github.com/tomek-szczesny/OpenUPS/files/11592459/OpenUPS_Diagram-v0.90.drawio.zip)

[image]
Source file: [OpenUPS_Diagram-v0.91.drawio.zip](https://github.com/tomek-szczesny/OpenUPS/files/11592465/OpenUPS_Diagram-v0.91.drawio.zip)

tomek-szczesny, 2023-05-29:
I was reluctant to work on a diagram because I haven't found any tool that beats pencil and paper, but I think I can work with that. :)

I'll expand the second diagram today - everything seems correct except one small detail, but also there are a few things missing. 

Feel free to add anything you like to the wiki. It is accessible as a separate repository: 
[git@github.com:tomek-szczesny/OpenUPS.wiki.git](git@github.com:tomek-szczesny/OpenUPS.wiki.git)
I'm not sure whether we can keep diagram in this repo, but that would be really helpful.

I recommend markdown editor, [marktext](https://github.com/marktext/marktext/releases/tag/v0.17.1). It explicitly supports github specific syntax if selected somewhere in the settings.

tomek-szczesny, 2023-05-29:
I don't think I instantly fell in love with this tool. Certainly better than many others I've seen before, but I'll have to invest some time learning it. Most notably, it doesn't seem to support branching arrows, which would be super useful to us.

I identified a reason for using a separate TWI for a battery - the BQ chip's I2C works at 100kHz, in contrast to usual 400kHz. We don't want to cap the whole bus like that. Also less traffic will improve latency on this critical bus.

There are still plenty of signals missing (mostly housekeeping metrics), but this is a good start for implementing necessary modules.

But to stay on topic, do you like the proposed battery handling solution?

[image]
[OpenUPS_Diagram-v0.92.drawio.zip](https://github.com/tomek-szczesny/OpenUPS/files/11593236/OpenUPS_Diagram-v0.92.drawio.zip)

hominoids, 2023-05-29:
I just started using drawio too and didn't take the time to find away to do branching arrows.  But I agree, it's better then most I have used.

The diagram looks good and is the best solution to date imo.  For discussion, there is a way to solve all the startup, safety, cable connection issues and also reduce the component count and cost significantly.  The BQ76200, ATtiny and all supporting components could be eliminated by simply moving the BQ76920 onto the UPS PCB.  By using the +Batt, -Batt and the balance/sense lines in a JST connector, the BQ76920 would know on startup or immediately while operating, if the battery pack or balance/sense lines were not attached and it would also know if someone removed any single battery.   The internal low-side FET of the BQ76920 could be used and the BQ76200 wouldn't be needed.  

For all of the above reason and maybe more, I suspect this is why most uses of multiple or different size packs use this method, it's safe.  It's very typical in some use categories for a battery pack to have power leads and a JST connector for the balance/sense lines.  It deserves some consideration and actually opens up supporting other battery packs, easily making your own and automatic battery configuration could be possible and be implemented in SW.  Just some things to thing about.  The UPS PCB could be widened back to 94mm adding 4mm more but I don't know if we could get any more depth, maybe 1mm-3mm at best but I would need to look.

tomek-szczesny, 2023-05-29:
I really do like this idea, with 8-pin JST cable this would work like you say, in nominal conditions. But I fear that won't solve all of the problems we've faced so far.

BQ76200 has been introduced because I saw no reasonable way of passing UART (or I2C) over the broken GND connection. This problem would still be there, unfortunately, and having it all on a single PCB won't get us any closer to a solution. 
High side switch would also be necessary to keep BQ76920 properly grounded at all times, otherwise it may disrupt the I2C operation when the battery is not present (pulling I2C lines up to charger's positive voltage, if not burning in the process). 

So, I guess this could work, with JST cable with at least 7 leads (6 for the cells, extra one for temperature sensor), but unfortunately with my current knowledge we're only saving attiny212.

We could however toy with an idea of replacing BQ76200 and high-side mosfets with a single-chip eFuse / hot swap solution. 
Such as [TPS25981](https://www.ti.com/lit/ds/symlink/tps25981.pdf). Except it's ridiculously small, but at least protects well against shorts and discharge overcurrent. No protection against charge current though.

tomek-szczesny, 2023-05-30:
Honestly I'm not convinced. That's a lot of effort to squeeze in all the battery related circuits into UPS mainboard. It may look empty now but most of that empty space is already dedicated to MOSFET cooling copper pours.
A much bigger JST connector will not make matters easier, not to mention rather unusual cable.

And I'm really worried how will BQ76920 react to battery voltage and sense signals appearing separately, in unspecified order. With high side switch that may be a minor issue, probably requiring clearing some error flags. 

In my opinion, this idea is worth further consideration only if we come up with a significant cost reduction of a battery related chipset. The most significant advantage of this approach is that the complexity of connections between battery support circuits and AVR is now unlimited.
I searched for a while, but it's unlikely we'll find a coulomb counter, battery monitor, balancer, and protection for a lower price.

For example, [BQ76925](https://www.ti.com/lit/ds/symlink/bq76925.pdf) is about half the price of BQ76920 but the main tradeoff is that it uses microcontroller's ADC. That can be arranged, AVR DB has "up to 22" ADC inputs. The aforementioned efuse is half the price of BQ76200 and associated FETs. I was unable to find a coulomb counter for multi-cell batteries that would not bring all these savings down to null.

There is also this affordable chip with everything we need. Too much in fact, also comes with 200 pages long programming manual and has an on-board CPU. The [BQ4050](https://www.ti.com/lit/ds/symlink/bq4050.pdf). 

You did mention that you are familiar with at least one instance of a battery management circuits separate from the battery pack - perhaps we could get some inspiration from them? Is there any open source project that does that?

tomek-szczesny, 2023-05-30:
L9961 has lost its appeal once I dug into its datasheet and realized the register map is nowhere to be found. HK had unpleasant experience with ST while developing SmartPower3 - they simply do not respond. So I won't even bother asking them for a missing piece of documentation, I suspect they withheld it for a reason.
Pity, it's a very cheap chip that does everything we wanted.

OK I did take a look on LibreSolar. Indeed, they are doing what you said they would, battery nanny on a separate board from the batteries.
The chips they support are super expensive, except our old pal BQ76920 and somewhat acceptable BQ76952.

It does everything we want. It is such a waste to employ a 15S capable chip in OpenUPS. There is a counterpart that works with only 10S packs, BQ76942, albeit the package is the same and price differs by mere 10%.
And you could benefit from the existing open source software, I reckon. This appears to be the most reasonable solution for now, but I'll have to take a closer look at this chip tomorrow.

It also came to my attention that we could use one of its three temperature monitoring channels to monitor UPS PCB temperature, instead of adding a separate I2C device for this purpose. Hooray, another dollar saved.

tomek-szczesny, 2023-06-02:
[image]
[OpenUPS_Diagram-v0.93.drawio.zip](https://github.com/tomek-szczesny/OpenUPS/files/11635262/OpenUPS_Diagram-v0.93.drawio.zip)

A slightly updated diagram with my freshest idea for dealing with battery pack.